riking/joycon #1: Support for Windows?

mfosse, 2017-03-28:
I'm able to get hidapi_demo.c working without much effort but the main program is a bit beyond me. In particular I can't seem to read any joystick data. Any help would be appreciated.

riking, 2017-03-28:
I set up the demo.c so if you press L or R it polls the joy-con for data, which should print out the stick values.

To get it working on Windows I need to replace the uinput output with windows output.

forerofore, 2017-03-28:
can you precompile it if you get windows support? i would love to test it (and i couldnt figure out how to compile the current source)

drew-wallace, 2017-06-06:
What if you tried compiling and using under Ubuntu for Windows?

riking, 2017-06-06:
Looks like a device driver is needed to provide input to other programs on Windows.

viperfx, 2017-10-03:
@riking just wondering what the status of this is? Is needing to create a device driver a big task? Just curious what the next steps are?

drew-wallace, 2017-10-03:
I might attempt to make a Universal Windows Driver with this code as a reference. I will be using the Joy-Cons as VR controllers and need the driver for the buttons and gyro input.