riking/joycon #2: Sync and get data without root

Philip-Scott, 2017-05-03:
Hey man! Great job with your tool :) 

Is it possible to get it working without having to run it with sudo? I would love to implement this on my "Powerpoint" like presentations app [Spice-Up](https://github.com/Philip-Scott/Spice-up/) :smile:

riking, 2017-05-03:
If you need just basic button-press functionality (not analog stick), you can just use the system-created gamepad device.

Philip-Scott, 2017-05-03:
To do that, do i need to point your code to another location, or should i just use a different tool? 

(Your's was actually the only one I've been able to get working)

riking, 2017-05-03:
No, you can just use the normal input APIs. If you connect one joy-con and go to http://html5gamepad.com/ it'll show up.

The goal of this program is to remap the buttons and combine the two joy-cons into a single input device.

Philip-Scott, 2017-05-03:
Thank you, now that i knew it works, i was able to use https://github.com/meghprkh/libgamepad but without mappings. Good luck :)

closed 2017-05-03 by Philip-Scott