riking/joycon #4: How do you plan on switching between modes?

drew-wallace, 2017-06-02:
Hey riking,

I was able to compile this on my Pi and it's working like a champ. I see you're re-writting this in Go. I'm hoping I will still be able to compile this on my pi once you're through. Currently, I'm using this driver for a pi project that basically mimics a Switch. I have a 7" screen, 3D printed case with built in Joy-Con rails on the sides, WiFi dongle, Bluetooth dongle, and soon it will be powered by a battery. It will primarily run a program called [Parsec](https://parsec.tv/) which is a super low latency desktop streaming solution for gaming. The goal is to be able to play full PC games on the go with as little latency as possible.

I am curious of your goals for this driver. Assuming it was left in it's current state, as a program that needs to be run, are you going to be adding any interesting features? Mainly, are you going to build in some button combination to allow for the coupling/decoupling of Joy-Cons? Or, is the program just for testing purposes as you write a native driver to be installed and never need to run anything?

Finally, thanks for making this. You are enabling one of my long time dreams of a homemade handheld console.

riking, 2017-06-05:
> Mainly, are you going to build in some button combination to allow for the coupling/decoupling of Joy-Cons?

Of course! L+R was always the intended button combination to make a controller, same as on the console. And the last few commits make it work as a uinput (userland input) driver.

Linux is easy because of uinput - you can just open() and write() events to it. Windows and Mac look like they need kernel driver development.

drew-wallace, 2017-06-05:
Now, am I going to need to compile this with Go or can I keep using the Makefile that targets your C code?

riking, 2017-06-05:
Nah, it's a cgo compile. I guess I'll update the Makefile.

I'm currently trying to figure out the stick calibration - it's not very usable when the stick values don't span the whole range.

drew-wallace, 2017-06-05:
On the calibration, you could make a program to guide you through it.

For automatic pairing and hardening, I had my joycons pair automatically with your build already. I also haven't had issues with them disconnecting. Is that something you've had issues with? If so, what environment are you in?

riking, 2017-06-05:
> On the calibration, you could make a program to guide the user through it.

Oh, no - the controllers **already have the calibration data on them.** I need to figure out the format of the data!

At the command line: `read u1 0x2024 0x1C` (or u2, c1, c2, etc, see `ls`) will dump the recalibration data (but not the factory-set calibration data, haven't found that yet).

drew-wallace, 2017-06-05:
Interesting. I'm not a hardware guy. It's something I'm slowly exposing myself to. In other words, I have no idea what that stuff means yet. I can test for you though. Thanks for the replies! If I get to a point where I'm of any use to you, I'll offer my help.

closed 2017-06-06 by drew-wallace