riking/joycon #5: SPI max read

CTCaer, 2017-06-06:
Is the maximum reported SPI data different between joy-cons and Pro controller?

I found out that 0x1D is the maximum from my joy-cons. 
Unfortunately, I don't own a pro controller to test

I implemented a async dump of both joy cons for https://github.com/shinyquagsire23/HID-Joy-Con-Whispering and fixed the extra bytes in dumps. Now I'm cleaning up stuff and would be good to know if this is the real maximum value for all devices, before I make a pull request.

riking, 2017-06-06:
Hmm, it was looking like the data was cutting off at 0x1C bytes for my JoyCons. Check the pointer math maybe? https://github.com/riking/joycon/commit/d62f019258179f8989e98c87fe309651129c683b#diff-763806487172f24f3200b6de69e882ebR570

CTCaer, 2017-06-06:
I don't understand GO language, but I will test your latest commit with my 4 joy cons after work.
And maybe try some changes to your code, to places that I understand what's going on.

The tests I did with a modified shinyquagsire23's code, were with a Blue left, Red right and 2 grey joycons.
First I used a safe length of 0x10. And got the 4 firmwares x3 times to make sure I got them correctly.
Then by checking the HID replies and the resulted dumps, I found out that the maximum data request for me was 0x1D. This is working for USB-HID and BT-HID.

(BTW, with a CSR 4.0 dongle, sometimes the BT connection halts for me and joy con starts to send 00 data replies. Maybe that's why for important big reads and writes, you must connect the joy con to switch).

riking, 2017-06-06:
If you pair it with L+R / SL+SR using my program, does it still occasionally disconnect?

CTCaer, 2017-06-10:
No. It keeps the connection pretty well.

Tested this many time with the following setup:
Jcmapper for initialization running in background
And shinyquagsire23's hidtest without joycons init. Just SPI dumping.

riking, 2018-01-16:
I think this was resolved by someone else.

closed 2018-01-16 by riking