riking/joycon #9: Mac support

riking, 2017-06-16:
- [x] Userspace input reading works fine
 - [ ] Cannot create a new input device without a **signed** kernel driver (or SIP off)

bryan1anderson, 2018-02-01:
any more progress on this?

riking, 2018-02-01:
It's basically just blocked on either (1) I acquire Mac hardware I have root on, sign up for Apple Developer program, learn how to write Mac kernel drivers, and get one signed or (2) Someone else does that

ahjsrhj, 2018-02-19:
so it means when I turn off SIP, it will works for other program?

CwbhX, 2018-06-25:
I'd be willing to try and work on the mac support but I don't have experience with Go, can't building the latest build due to undefined "sync" in bluetooth_generic. If someone has a fast answer that'd be great since not much time to debug after work. cheers

riking, 2018-07-03:
@CwbhX whoops! just running `goimports` will fix that one, will push up a commit right now.

It's hard to test on all OSes because cgo means I need a builder on the target system, or a full cross compiler (not happening).

f9c1217

nickell-andrew, 2018-07-06:
Is there a work around for mac OS setup regarding libudev-dev? I'm trying to get this whole thing to compile on my macbook pro.

gerd2002, 2018-07-15:
Joycon by itself works fine under macOS. The issue is that you cannot actually create an input driver in userspace. @nickell-andrew 

@riking I recently (don't ask me how, I don't remember) discovered unbit/foohid, which apparently allows you to write an input driver in userspace. I don't think of myself as experienced enough to write a good input driver, but I'll try anyways.

nickell-andrew, 2018-07-24:
@gerd2002 nice find on the unbit/foohid
I may take a stab at writing a driver myself even though I have no experience writing input drivers. Feel free to bother me if you think I can help you in any way with you efforts.