riking/joycon #12: Trying to get this to work with SDL

drew-wallace, 2017-06-21:
Your C driver works with SDL, but the Go version doesn't. Not sure why...

riking, 2017-06-27:
Interesting. I think it has to do with which button outputs are defined - it might not like the TRIGGER_HAPPY ("extra buttons") outputs.

drew-wallace, 2017-06-27:
I literally just figured it out right now. Would you like a PR?

drew-wallace, 2017-06-28:
From this
```
{"GamepadCapture", C.BTN_TASK},
```
Back to this
```
{"GamepadCapture", C.KEY_SYSRQ},
```

And this
```
var linuxAxisNames = []linuxKeyCode{
    // have to avoid ABS_X and ABS_Y because Xorg thinks it's a mouse despite BTN_GAMEPAD
    {"MainStickHoriz", C.ABS_Z},
    {"MainStickVertical", C.ABS_RX},
    {"SecondStickHoriz", C.ABS_RY},
    {"SecondStickVertical", C.ABS_RZ},
}
```
to this
```
var linuxAxisNames = []linuxKeyCode{
    // have to avoid ABS_X and ABS_Y because Xorg thinks it's a mouse despite BTN_GAMEPAD
    {"MainStickHoriz", C.ABS_X},
    {"MainStickVertical", C.ABS_Y},
    {"SecondStickHoriz", C.ABS_RX},
    {"SecondStickVertical", C.ABS_RY},
}
```
Also, the horizontal axis on the main sick of the full joycon mapping needs to be set as inverted in order for it to be normal in game. In every game I tested, left and right were switched, only for the full controller. The individual ones were fine.

drew-wallace, 2017-06-28:
Also, if it helps, setting the Capture button to `C.BTN_TASK` made SDL think my joycon was a mouse, and you must have `ABS_X` and `ABS_Y` set so it thinks it's a joystick.

riking, 2017-06-30:
Wow. Lots of conflicting requirements…

riking, 2017-06-30:
Hmm, what happens if we register axis X and Y without using them?

drew-wallace, 2017-06-30:
That would be fine. I'd have to change the SDL mapping to point at the right axis index for each stick.

closed 2017-07-23 by drew-wallace