riking/joycon #15: [RFC] Retrieve analog sticks factory and user calibration

CTCaer, 2017-08-24:
Hey,

check the latest 2 commits:
https://github.com/CTCaer/jc_toolkit/commits/master

It's preliminary proof of concept on how to parse the calibration for the analog sticks.
The data includes center position and x/y distance from center.

Lastly, I found out, that the factory calibration is used by switch, if no user calibration.
I changed the center position and the position on screen was wrong.

riking, 2017-09-01:
Yup, if you look I already added a handler for SPI reads: https://github.com/riking/joycon/blob/master/prog4/joycon/bluetooth.go#L606

CTCaer, 2017-09-02:
yeah don't worry. It was before I wrote the documentation in dekunukem's repository.
(And I had some bugs, so the commits on 24-27 august are my controls test)

riking, 2017-09-05:
hey don't close it the issue isn't actually fixed

PR #17 (2017-10-24): Implement joystick calibration from device

 - Fix battery to separate out levels from charging
 - Rename input modes
 - Add `decodeUint12` function
 - Change stick data to int16, change output range to (-0x7FF, +0x7FF)
 - Change stick calibration function

Fixes #15, #16
closed 2017-10-24 by riking