riking/joycon #18: Control sticks inverted

Noahkiq, 2017-12-29:
Recently found this project and got it setup to work in a few programs. While setting up my JoyCons in antimicro, I noticed that left=right, up=down, etc. This wasn't an issue in most programs I tried as they let me specify which direction should be positive/negative (ie retroarch, which has you setup control sticks by pressing up, down, left, and right). However, Steam only detects which axises are up/down and left/right, meaning all directions are inverted in Steam games, and I've found no option in Steam to fix this. Any help fixing this would be greatly appreciated 😄

Noahkiq, 2018-01-05:
Tried it with `jcmapper` instead after realizing I didn't build that, and it fixed the horizontal axis on both sticks, but not the vertical axis.

riking, 2018-01-16:
Button remapping is on the to-do list. Take a look at the "invert" field in this file: https://github.com/riking/joycon/blob/master/prog4/output/keymap_common.go

Noahkiq, 2018-01-17:
Swapping the `true`s and `false`s fixes the issue for me, thanks :)

PR #33 (2018-06-06): Axis-remapping via cli-flags
Fixes #18 .

You can now specify --invert (L/R)(V/H) when running jcdriver to invert a specific axis.

I had to modify a whole bunch of files to properly pass the options to the input-processing.
It basically boils down to:
- interface.go : Now containes the new Options-Struct
- main.go : Specifies and parses cli-Options into an Options-Struct
- Manager.go : New() receives an Options struct. The Input-Remapping part of the options is passed on to the output factory
- output_uinput.go : passes the remapping-options to the instantiated uinput thingy
- output_linux.go [... cut]
closed 2018-06-06 by riking