riking/joycon #19: Documentation?

arm32x, 2018-01-05:
I would like to use this program but I don't know how to compile it. Could you please make some documentation on that?

Noahkiq, 2018-01-05:
I used one of @drew-wallace's [switchberry scripts](https://github.com/drew-wallace/switchberry/blob/master/4-joycons) (for ubuntu/debian, probably not too hard to edit for other distributions). It should output a `jcdriver` file to `~/go/bin`. Just connect your joycons with bluetooth, run `sudo ./jcdriver`, and pair your joycons by pressing L + R.

(Well, I actually used his script with some slight modifications that actually made it work on my computer. [link to that too](https://raw.githubusercontent.com/Noahkiq/switchberry/master/4-joycons))

riking, 2018-01-16:
I could produce binary releases at some point if there's interest.

nickell-andrew, 2018-07-24:
@arm32x: @riking recently merged my PR to update the setup instructions to make them a bit more noob friendly (like myself). If you try again and let me know what issues you encounter I can help you debug them and improve the docs at the same time.