riking/joycon #21: [Question] Is this usable without Switch?

TsundereBug, 2018-01-08:
You said calibrated JoyCons were needed, however my JoyCons are fresh out of the box and as far as I know have not been connected to a switch.

Do I just connect the joycons via Bluetooth? Do they work together as 1 single controller?

Noahkiq, 2018-01-09:
yeah, the switch isn't needed at all. just connect the joycons to your PC via bluetooth and run `jcdriver` (which will merge them into one input). if you haven't compiled it yet, [you can use this script for ubuntu](https://raw.githubusercontent.com/Noahkiq/switchberry/master/4-joycons)

you can also build `jcmapper`, which (iirc?) only requires `gcc`. i think the difference is that `jcmapper` is supposed to let you do your own control stick calibration (but i haven't gotten that to work)

TsundereBug, 2018-01-10:
I am on Fedora and have successfully compiled the entire source.

Unfortunately, my bluetooth is broken, but I posted a help thread on [FedoraForum](https://forums.fedoraforum.org/showthread.php?316857-Bluetooth-stuck-in-off&p=1801357#post1801357).

riking, 2018-01-16:
The factory calibration should work fine on the latest code!

closed 2018-06-13 by sorenbug