riking/joycon #23: Need Help!

M3meZ, 2018-01-16:
make
go install -v ./prog4/jcdriver
make: go: Command not found
Makefile:34: recipe for target 'jcdriver' failed
make: *** [jcdriver] Error 127
 
Any Solutions Anyone?

M3meZ, 2018-01-16:
Fixed It But Theres a New Error ._.

make
go install -v ./prog4/jcdriver
prog4/jcdriver/main.go:8:2: cannot find package "github.com/riking/joycon/prog4/consoleiface" in any of:
	/usr/local/go/src/github.com/riking/joycon/prog4/consoleiface (from $GOROOT)
	/home/vroom/Projects/Proj1/src/github.com/riking/joycon/prog4/consoleiface (from $GOPATH)
prog4/jcdriver/factory_uinput.go:8:2: cannot find package "github.com/riking/joycon/prog4/jcpc" in any of:
	/usr/local/go/src/github.com/riking/joycon/prog4/jcpc (from $GOROOT)
	/home/vroom/Projects/Proj1/src/github.com/riking/joycon/prog4/jcpc (from $GOPATH)
prog4/jcdriver/factory_uinput.go:9:2: cannot find package "github.com/riking/joycon/prog4/output" in any of:
	/usr/local/go/src/github.com/riking/joycon/prog4/output (from $GOROOT)
	/home/vroom/Projects/Proj1/src/github.com/riking/joycon/prog4/output (from $GOPATH)
Makefile:34: recipe for target 'jcdriver' failed
make: *** [jcdriver] Error 1

Noahkiq, 2018-01-16:
Try using @drew-wallace's [switchberry script](https://github.com/drew-wallace/switchberry/blob/master/4-joycons)! It's (as the name implies) for Raspberry Pis, but works fine on Ubuntu (with minor tweaking). You can also easily find the packages in the `apt-get install` to install on other distributions.

M3meZ, 2018-01-16:
The script does not work and would always end up in errors i even changed directories but still no luck

drew-wallace, 2018-01-16:
I hope you didn't try to use the script as is...

Jayfire, 2018-02-17:
How did you fix the original error?