riking/joycon #24: UI to re-pair or shuffle around controllers

riking, 2018-01-16:
There should be an optional graphical UI to disconnect and/or re-pair the controllers from paired into single mode.

We could trigger this from the HOME button

djwraith, 2018-02-01:
THIS WOULD BE AMAZING!!
Just implemented your work in my RetroPie project.  Trying to figure out my own solution to easily swap between modes but I'm definitely a novice.  Watching your work closely!

UsernameNotAvalible, 2020-11-06:
Maybe someone could add the ability to be able to bind controller buttons to act like other buttons, maybe even keyboard keys such as printscreen. then, you could have a special bind option so people could bind the home button(or any button!) to the menu. Just an idea. i also think it would be cool if it opened the GUI by default, but still let you use the terminal interface.