riking/joycon #25: Controller bindings

Xobtah, 2018-01-25:
Hi!

I'm playing Steam games with the paired Joy-Cons controller, but I have a problem with the controller bindings. Steam thinks the controller is an XBox controller, and thus swaps buttons A <-> B & Y <-> X. Moreover, the X and Y axis on both joysticks are inverted, for instance giving the top right direction on the controller will make the character go down left.

Is there a handy way to edit those bindings?

Thank you for your time!

Noahkiq, 2018-02-10:
Fixing the button issue is as simple as just manually swapping them yourself in Steam's controller settings. Fixing the inverted control stick axis involves editing some files and recompiling (see #18), or you could use [my fork](https://github.com/Noahkiq/joycon/) which does that for you.

Xobtah, 2018-02-10:
I have tried with various games and it works perfectly on Steam Linux, thanks a lot!

closed 2018-02-10 by Xobtah