riking/joycon #26: panic: runtime error: integer divide by zero

frntz, 2018-02-07:
When I'm trying to push L+R I have this issue
```
plonk
plonk
pairing double
panic: runtime error: integer divide by zero

goroutine 19 [running]:
github.com/riking/joycon/prog4/joycon.(*calibrationData).Adjust(0x105fa1bc, 0x8e00763, 0x0)
	/home/pi/go/src/github.com/riking/joycon/prog4/joycon/common.go:98 +0x3b8
github.com/riking/joycon/prog4/joycon.(*joyconBluetooth).ReadInto(0x105fa180, 0x10612b10, 0x1d9200)
	/home/pi/go/src/github.com/riking/joycon/prog4/joycon/bluetooth.go:153 +0x228
github.com/riking/joycon/prog4/controller.(*two).JoyConUpdate(0x10612b00, 0x1d92a0, 0x105fa180, 0x1)
	/home/pi/go/src/github.com/riking/joycon/prog4/controller/two.go:47 +0x120
github.com/riking/joycon/prog4/joycon.notify(0x1d92a0, 0x105fa180, 0x1, 0x1052efd4, 0x2, 0x2)
	/home/pi/go/src/github.com/riking/joycon/prog4/joycon/common.go:15 +0x6c
github.com/riking/joycon/prog4/joycon.(*joyconBluetooth).reader(0x105fa180)
	/home/pi/go/src/github.com/riking/joycon/prog4/joycon/bluetooth.go:570 +0x410
created by github.com/riking/joycon/prog4/joycon.NewBluetooth
	/home/pi/go/src/github.com/riking/joycon/prog4/joycon/bluetooth.go:71 +0xdc
```

Noahkiq, 2018-02-07:
I've had this issue too, but only after connecting joycons after I've started `jcdriver`. Connecting my joycons then running jcdriver fixes this issue. Could be related to another issue I've noticed where the user stick calibration defaulted to all zeroes when connecting my right joycon after jcdriver.

frntz, 2018-02-08:
thanks @Noahkiq ! it's fixed thanks to your tips ! Error happens some times when the calibration is at zero as well ! I notice some lags while playing left joy con is almost playable but the right one it's impossible to play ... (It may be a Retroarch issue / configuration thing)

commit 12c642f4 (2018-02-09): Fix divide-by-zero error on null calibrations

Fixes #26
closed 2018-02-09 by riking