riking/joycon #30: Retropie

Teeqla, 2018-03-13:
Can someone please explain how I can set this up on Retropie.
I'm new and can't figure it out.
Thanks!

TsundereBug, 2018-03-14:
From reading, Retropie is debian.
```
sudo apt install go git # install GoLang and Git
git clone https://github.com/riking/joycon.git # Clone this repo
cd joycon # Enter the repo
go get github.com/riking/joycon/prog4/consoleiface # Get the required dependencies
mkdir bin # Make a bin folder because Go
export GOBIN=$PWD/bin # Set the GOBIN folder
make -j4 # Make on 4 threads (set to -j2 for 2 threads, or no argument for 1 thread)
sudo ./jcdriver # Run the driver
```
Once you have that running, connect your joycons, and sync them like you would sync them to a switch (L and R).

Teeqla, 2018-03-30:
How to set the $GOPATH??
It says always: 
go get github.com/riking/joycon/prog4/consoleiface

package github.com/riking/joycon/prog4/consoleiface: cannot download, $GOPATH not set. For more details see: go help gopath

Julius-Beides, 2018-06-04:
Can this script from @sorenbug  be put into the readme.md?
@Teeqla It was working fine for me (Ubuntu 18.04)

riking, 2018-06-04:
@Julius-Beides Thanks for the suggestion, just updated the README in https://github.com/riking/joycon/commit/4ea5386aa095ced1c90f10a1171279aa3f0e46fe .

JPHanke, 2018-12-21:
@sorenbug thanks for the instructions! It worked fine. But how do I go back to the emulationstation with the driver running?

sorenbug, 2018-12-22:
@Teeqla `export GOPATH=[the directory you want as $GOPATH]`, I think I remember using either `~/go` or `$PWD` for it, can't remember. Someone who actually uses Go would probably be able to help you better.

@JPHanke Retropie runs in a specific `tty`, right? If you press <kbd>Ctrl</kbd>+<kbd>Alt</kbd>+<kbd>F2</kbd> with a keyboard plugged in, it should open you to another screen where you can log in and run commands apart from Retropie.

JPHanke, 2018-12-22:
It worked, thank you!!

Toothless2005, 2019-01-03:
when I do make -j4 it gives me this:
go install -v ./prog4/jcdriver
github.com/riking/joycon/vendor/github.com/GeertJohan/go.hid
# github.com/riking/joycon/vendor/github.com/GeertJohan/go.hid
../src/github.com/riking/joycon/vendor/github.com/GeertJohan/go.hid/hidraw_linux.go:265: undefined: os.ErrClosed
../src/github.com/riking/joycon/vendor/github.com/GeertJohan/go.hid/hidraw_linux.go:303: undefined: os.ErrClosed
../src/github.com/riking/joycon/vendor/github.com/GeertJohan/go.hid/hidraw_linux.go:341: undefined: os.ErrClosed
../src/github.com/riking/joycon/vendor/github.com/GeertJohan/go.hid/hidraw_linux.go:386: undefined: os.ErrClosed
../src/github.com/riking/joycon/vendor/github.com/GeertJohan/go.hid/hidraw_linux.go:415: undefined: os.ErrClosed
../src/github.com/riking/joycon/vendor/github.com/GeertJohan/go.hid/hidraw_linux.go:479: undefined: os.ErrClosed
Makefile:34: recipe for target 'jcdriver' failed
make: *** [jcdriver] Error 2

Toothless2005, 2019-01-03:
when I do make -j4 it gives me this message:

go install -v ./prog4/jcdriver
github.com/riking/joycon/vendor/github.com/GeertJohan/go.hid
# github.com/riking/joycon/vendor/github.com/GeertJohan/go.hid
../src/github.com/riking/joycon/vendor/github.com/GeertJohan/go.hid/hidraw_linux.go:265: undefined: os.ErrClosed
../src/github.com/riking/joycon/vendor/github.com/GeertJohan/go.hid/hidraw_linux.go:303: undefined: os.ErrClosed
../src/github.com/riking/joycon/vendor/github.com/GeertJohan/go.hid/hidraw_linux.go:341: undefined: os.ErrClosed
../src/github.com/riking/joycon/vendor/github.com/GeertJohan/go.hid/hidraw_linux.go:386: undefined: os.ErrClosed
../src/github.com/riking/joycon/vendor/github.com/GeertJohan/go.hid/hidraw_linux.go:415: undefined: os.ErrClosed
../src/github.com/riking/joycon/vendor/github.com/GeertJohan/go.hid/hidraw_linux.go:479: undefined: os.ErrClosed
Makefile:34: recipe for target 'jcdriver' failed
make: *** [jcdriver] Error 2

sorenbug, 2019-01-04:
@Toothless2005 most likely an issue with the version of Go you're using.

Davidc999, 2020-10-25:
Does @sorenbug s comment assume I'm running retropie in a desktop environment? I've setup my pi to launch retropie on boot and have no desktop environment. Any way to keep the driver running and launch retropie from command prompt?

Davidc999, 2020-10-25:
> Does @sorenbug s comment assume I'm running retropie in a desktop environment? I've setup my pi to launch retropie on boot and have no desktop environment. Any way to keep the driver running and launch retropie from command prompt?

Ah, I solved it by using the screen app