riking/joycon #35: Not working on latest clone

sorenbug, 2018-06-09:
Followed instructions in readme and joycon does not connect as an input device or controller (tested via http://html5gamepad.com/)

sorenbug, 2018-06-09:
> are you pressing L+R?

Yes

sorenbug, 2018-07-14:
Switching to the `develop` branch also does not seem to work. I run `sync`, hold the sync button on my JoyCon, and nothing happens.

Noahkiq, 2018-07-14:
Is `jcdriver` running? Are you actually connecting to them via your system's Bluetooth settings? And are you pressing L + R on your JoyCons after `jcdriver` confirms your JoyCons have connected?

sorenbug, 2018-07-14:
jcdriver tells me to scan using it and not connect externally. When I do
connect externally, it is not registered in jcdriver. Pressing L and R do
nothing at any point.

riking, 2018-07-14:
Yeah, the auto connection is the feature I was working on, it was broken, so the changes I made were to disable it on `master`.

riking, 2018-07-14:
Also confirm that you're running on Linux? I haven't written an OS X kernel driver yet, so it can't add a device on Mac.

sorenbug, 2018-07-14:
I am running on Arch Linux under i3wm and managing devices through
blueman-manager.

sorenbug, 2018-07-15:
I hold the sync button, pair through blueman-manager, and nothing happens in the console or with my joycon's player lights (even when pressing L+R). The joycon disconnects after a time.

sorenbug, 2018-07-15:
YES ok so, connecting it via the HID service connects it to the driver. I'll try connecting a second joycon.

riking, 2018-07-16:
@sorenbug that's probably a bug with your bluetooth card / driver then :( You'll probably need to buy a different BT radio.

sorenbug, 2018-07-19:
Which is weird, because using mfosse's driver for windows works fine. I think it's a bug with linux bluez that nobody has bothered to resolve/look into yet.