riking/joycon #39: Controller not recognized by games

frankruegamer, 2018-07-21:
I can connect both Joy-Cons just fine, so that both of them have one steady light. But after this I don't know how to proceed, it wasn't working in any game I've tried so far, though some games (i.e. Enter the Gungeon) showed that there was a controller connected.

Am I missing something?

nickell-andrew, 2018-07-22:
Can you give more information about your setup? Are you using this with OSX, Windows, Linux? Which flavor/version?

joycon's functionality in OSX is little more than a proof of concept as no one has written a driver to expose the joined joycons as a single controller to other software on an OSX system.

frankruegamer, 2018-07-23:
I'm using Arch Linux. To add to that, testing tools like jstest appear to be able to detect buttons pressed and sticks moved (at least with the left joy con).

nickell-andrew, 2018-07-24:
I don't know how helpful my input will be then @frankruegamer . I've only played around on OSX.

While I can't speak to integration inside games, I can say that the driver when running in a terminal successfully created a composite controller after I had both controllers connected via bluetooth.

Are you sure you have both joy con's successfully connected individually?

riking, 2018-07-29:
What does html5gamepad.com show? Sometimes, it's just that the game has weird detection rules and you need to tweak the output config per game (another reason to have actual config files).

frankruegamer, 2018-07-29:
This site is able to detect every input (except gyro and screenshot) from my joycons. I tried it in multiple games (including Dead Cells, Enter the Gungeon, Tomb Rider, Wizard of Legend, Hyper Light Drifter, ...); while most of them were able to detect a controller, none would accept input from them. Two of them, Tomb Rider and Hyper Light Drifter, detected a fast and constant movement of a stick so it would switch menu entries really quickly. They work just fine with my PS3 controller.

riking, 2018-07-31:
Yeah, so that means they don't like the output profile of the controller. Unfortunately, I don't really have time right now to work on creating a configuration file for that :(

JPHanke, 2018-12-22:
Hi all, I'm facing the same issue... I can configure the joy cons as one single joystick in my retropie, but the games don't recognize it. Is there a solution for it? Or can you give us some directions so we can create our own config file? Thanks.

Davidc999, 2020-11-10:
Your issue with retropie is probably that it detects 3 controllers: L, R and the full joycon. You have to rearrange the controllers in the config so that the full controller is player one, then it should work