riking/joycon #45: Only 1 connects

jorpilo, 2018-08-16:
Hey, I am trying to set this up and when connecting only 1 works perfectly but when trying to connect the second one it doesn't pick it up, my bluethoot connects it but the program doesnt show anything,

Am I doing something wrong?

It works independenly of the controles, so no matter if L or R only the first one is picked up

dbaumgarten, 2019-03-01:
This does happen to me sometimes.
Usually unpairing and repairing both joycons fixes this.
But I still have no idea why exactly this happens.

mid-kid, 2019-04-07:
Same problem here, except unpairing and repairing doesn't solve anything.
Connecting one joycon independently works flawlessly, connecting two of them, however, just causes the second connection to hang, and eventually give an "Input/Output" error. This happens regardless of whether the program is running. The joycons are on firmware 6.2.0 and they have been disconnected from the console.

mid-kid, 2019-04-07:
Just tested out a USB dongle. It appears that using this dongle, I can get both to connect, which leads me to believe that there's something wrong with the driver or the Bluetooth device on my Thinkpad A485.
The dongle I used is a "Ninser Bluetooth 4.0 USB Adapter", `Bus 002 Device 007: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)`, I have no clue what the internal bluetooth device is called.