riking/joycon #48: The program crashes on start on Arch Linux

aldelaro5, 2018-09-20:
I installed the aur package, entered the command `sudo jcdriver` and I get this:

```[console]> panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0xb8 pc=0x503a2c]

goroutine 1 [running]:
github.com/riking/joycon/prog4/consoleiface.(*Manager).SearchDevices(0xc000086640, 0x0, 0x0)
	/tmp/yaourt-tmp-aldelaro5/aur-joycon-git/src/joycon/prog4/jcdriver/src/github.com/riking/joycon/prog4/consoleiface/Manager.go:448 +0x31c
github.com/riking/joycon/prog4/consoleiface.(*Manager).Run(0xc000086640)
	/tmp/yaourt-tmp-aldelaro5/aur-joycon-git/src/joycon/prog4/jcdriver/src/github.com/riking/joycon/prog4/consoleiface/Manager.go:77 +0x41e
main.main()
	/tmp/yaourt-tmp-aldelaro5/aur-joycon-git/src/joycon/prog4/jcdriver/main.go:49 +0x2c4
```

Is this expected or have I done something wrong?

parkerlreed, 2018-09-20:
Working fine here on Arch

````
[parker@yoga710 ~]$ sudo jcdriver 
[console]> [INFO] Connected to Joy-Con L 58:2f:40:2d:3e:bc
Joy-Con L (58:2f:40:2d:3e:bc): 🔋 █ 
58:2f:40:2d:3e:bc: Got factory calibration and button colors
58:2f:40:2d:3e:bc: SPI read returned [603d+25]
00000000  d7 84 3c a9 57 8e 88 c4  46 ff ff ff ff ff ff ff  |..<.W...F.......|
00000010  ff ff ff 82 82 82 0f 0f  0f                       |.........|
58:2f:40:2d:3e:bc: SPI read returned [8010+22]
00000000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000010  ff ff ff ff ff ff                                 |......|
58:2f:40:2d:3e:bc: Checked user stick calibration: [{1239 968 1961 2277 1160 1132} {4095 4095 4095 4095 4095 4095}]
[INFO] Connected to Joy-Con R 58:2f:40:32:78:2c
Joy-Con R (58:2f:40:32:78:2c): 🔋 █ 
58:2f:40:32:78:2c: Got factory calibration and button colors
58:2f:40:32:78:2c: SPI read returned [603d+25]
00000000  ff ff ff ff ff ff ff ff  ff 31 28 76 e9 84 45 49  |.........1(v..EI|
00000010  95 43 ff 82 82 82 0f 0f  0f                       |.C.......|
58:2f:40:32:78:2c: SPI read returned [8010+22]
00000000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000010  ff ff ff ff ff ff                                 |......|
58:2f:40:32:78:2c: Checked user stick calibration: [{4095 4095 4095 4095 4095 4095} {1353 1081 2097 1890 1257 1112}]
Plonk! (u0)
Plonk! (u1)
pairing double
set permissions for js3
set permissions for event20
````

aldelaro5, 2018-09-20:
so I eventually got it to run, the problem is I rebooted and then it made the same error, I have no idea how to troubleshoot this one

aldelaro5, 2018-09-21:
Okay so this issue is much more clear now as I was very determined to make this work so I tried with a debugger and what I found is that in manager.go, the search device function, it has a switch case with the product id and it doesn't have a default case.  This is a problem because say you have another nintendo device like a wiiu pro controller paired (it doesn't even need to be paired, just existing on my end was enough for the program to pick it up), it would assume it was a joycon, but because it wasn't, the information about it aren't initialized so it crash when it tries to print them.

I solved it by simply adding a default case and just do a `continue` in it.  Idk if this is the correct solution, but this hack worked for me.  There essentially needs to have a way to not consider any nintendo device that aren't joycons.

riking, 2018-10-07:
Oh, yeah that'd do it...