riking/joycon #52: Unity games crash when controller is connected

samjhandley, 2018-11-19:
There is a strange issue I've come across that I can't really nail down. It could be a Unity(game engine), Steam, Linux or JCdriver issue.

Using Steam on Arch Linux, If I have the joycons paired and in single controller mode, setup through Steam as a Switch Pro controller and launch a Unity game it will crash. Disconnect the controller and it launches no problem. Controller works great with Steam and seems to be detected as an Xbox controller in non-Unity games. 

Happy to provide some outputs to stuff if anyone has some ideas of where to start.

lexikiq, 2020-01-13:
Have reproduced on Ultimate Chicken Horse and Robot Roller-Derby Disco Dodgeball. Contacted developers of the former about this once believing it was a game issue and they replied

> From the log, it seems like Unity is having problems dealing with the Joycons, which means that unfortunately there's not a lot we can likely do about it.

lexikiq, 2020-01-13:
The logs show a crash relating to the name of the joycons somehow. Not sure how "Full Joy-Con 1" would be crashing it, unless it's missing the name somehow or something. Closest related issue is [here](https://issuetracker.unity3d.com/issues/linux-crash-in-unityengine-dot-input-dot-getjoysticknames-when-controllers-with-comma-in-name-are-plugged-in)

georgefst, 2020-08-05:
Did you ever make any progress?

I've been looking in to this after finding that [my virtual gamepad](https://github.com/georgefst/monpad) crashes Ultimate Chicken Horse at startup.

AFAICT, Unity's handling of gamepads on Linux is just awful. In my experience, games just crash at startup with anything other than a PS or Xbox controller, without anything useful in the logs.

I'd be interested if anyone has any evidence to the contrary...