riking/joycon #53: Build issues on OSX 10.14.1

lotophagus, 2018-11-30:
Trying to run the code on OSX, I get the following error when downloading and building from GitHub:

`$ go get -u github.com/riking/joycon/prog4/jcdriver
# github.com/riking/joycon/prog4/jcdriver
go/src/github.com/riking/joycon/prog4/jcdriver/output_generic.go:11:54: undefined: InputRemappingOptions`

When I try to manually build the file, I get the same error:

`$ go build 'github.com'/riking/joycon/prog4/jcdriver
# github.com/riking/joycon/prog4/jcdriver
jcdriver/output_generic.go:11:54: undefined: InputRemappingOptions`

Brand new to Golang, been tinkering around with it which is how I figured out how to manually build, but I can't find any information through google on this error, which leads me to believe there's something I'm missing here?

Any ideas? thanks!

Noahkiq, 2018-11-30:
this isn't supported on mac

uclatommy, 2019-01-23:
@durteetr, You will have to check out a version before breaking changes were introduced. (These are the axis remapping updates made by @dbaumgarten). First, follow the instructions as written.
```
$ brew install go
$ go get -u github.com/riking/joycon/prog4/jcdriver
```
And you will get the following compile error:
```
# github.com/riking/joycon/prog4/jcdriver
go/src/github.com/riking/joycon/prog4/jcdriver/output_generic.go:11:54: undefined: InputRemappingOptions
```
In order to fix this, we will need to (1) go to the source directory created by that last command we used and (2) revert the checked out files to an earlier version:
```
$ cd go/src/github.com/riking/joycon
$ git checkout acba05838b16598ebd5fa75296b30b8d81a9cb64
```
Before you can compile this, there is one commit that fixes a compile error that you will need to cherry pick:
```
$ git checkout f9c1217ca527d5fb553816d8371c762bb335d9c9 -- prog4/jcdriver/bluetooth_generic.go
```
Now you can manually compile:
```
$ go build github.com/riking/joycon/prog4/jcdriver
```
And then run the driver:
```
$ ./jcdriver
```
Note that even though the controllers will be recognized in the terminal window and all button presses will correctly register and display output, no one has written any way to expose this to osx as a gamepad.
```
[console]> [INFO] Connected to Joy-Con R a4-38-cc-ba-63-00
[INFO] Connected to Joy-Con L a4-38-cc-b7-a9-b1
Joy-Con R (a4-38-cc-ba-63-00): 🔋 ▃ 
Joy-Con L (a4-38-cc-b7-a9-b1): 🔋 ▅ 
a4-38-cc-ba-63-00: Got factory calibration and button colors
a4-38-cc-ba-63-00: SPI read returned [603d+25]
00000000  ff ff ff ff ff ff ff ff  ff c6 68 78 49 95 43 46  |..........hxI.CF|
00000010  f5 42 ff 82 82 82 0f 0f  0f                       |.B.......|
a4-38-cc-b7-a9-b1: Got factory calibration and button colors
a4-38-cc-b7-a9-b1: SPI read returned [603d+25]
00000000  c4 55 44 46 67 90 50 65  47 ff ff ff ff ff ff ff  |.UDFg.PeG.......|
00000010  ff ff ff 82 82 82 0f 0f  0f                       |.........|
a4-38-cc-b7-a9-b1: SPI read returned [8010+22]
00000000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000010  ff ff ff ff ff ff                                 |......|
a4-38-cc-b7-a9-b1: Checked user stick calibration: [{1476 1093 1862 2310 1360 1142} {4095 4095 4095 4095 4095 4095}]
a4-38-cc-ba-63-00: SPI read returned [8010+22]
00000000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000010  ff ff ff ff ff ff                                 |......|
a4-38-cc-ba-63-00: Checked user stick calibration: [{4095 4095 4095 4095 4095 4095} {1350 1071 2246 1926 1353 1081}]
Plonk! (u0)
Plonk! (u1)
pairing double
[Controller 1] R pressed
[Controller 1] R released
[Controller 1] ZL pressed
[Controller 1] ZL released
[Controller 1] ZR pressed
[Controller 1] ZR released
[Controller 1] Right pressed
[Controller 1] Right released
[Controller 1] Home pressed
[Controller 1] Home released
[Controller 1] Down pressed
[Controller 1] Down released
```

dbaumgarten, 2019-03-01:
I tired to fix whatever I broke there, but when trying to cross-compile the program for macOS on my ubuntu i got several errors from a vendored library:

```
# github.com/riking/joycon/vendor/github.com/GeertJohan/cgo.wchar
../../vendor/github.com/GeertJohan/cgo.wchar/wchar.go:15:12: undefined: convertWcharToGoRune
../../vendor/github.com/GeertJohan/cgo.wchar/wchar.go:23:9: undefined: convertGoRuneToWchar
../../vendor/github.com/GeertJohan/cgo.wchar/wchar.go:47:9: undefined: convertGoStringToWcharString
../../vendor/github.com/GeertJohan/cgo.wchar/wchar.go:128:14: undefined: convertWcharStringToGoString
../../vendor/github.com/GeertJohan/cgo.wchar/wchar.go:141:9: undefined: convertWcharStringToGoString
../../vendor/github.com/GeertJohan/cgo.wchar/wchar.go:150:9: undefined: convertWcharStringToGoString
../../vendor/github.com/GeertJohan/cgo.wchar/wchar.go:158:9: undefined: convertWcharToGoRune
```

Seems like cross-compiling is broken here for mac. And as I don't have a mac to test thing there is not really much i can do here.

Also the readme states that the program is not really working on mac. It correctly registers the pressed buttons, but can not forward the button presses to other applications, as this would require a kernel module.