riking/joycon #54: Compilation keeps failing

4techguns, 2019-01-04:
Every time I try to run go get -u github.com/riking/joycon/prog4/jcdriver it shows this:
`~$ go get -u github.com/riking/joycon/prog4/jcdriver # Download and build the program

# github.com/riking/joycon/vendor/github.com/jochenvg/go-udev
go/src/github.com/riking/joycon/vendor/github.com/jochenvg/go-udev/device.go:7:12: fatal error: libudev.h: No such file or directory
   #include <libudev.h>
            ^~~~~~~~~~~
compilation terminated.
`
and then exits.

parkerlreed, 2019-01-04:
You need to install that library. Depending on distro, it's probably `libudev-dev`

4techguns, 2019-01-10:
[image]
Nope. Did not work.

parkerlreed, 2019-01-10:
What distro are you on? I only gave debian based as an example.

4techguns, 2019-01-10:
Fedora

4techguns, 2019-01-10:
Nevermind, the lib was in systemd-devel.

closed 2019-01-10 by 4techguns