riking/joycon #64: connected but can't detect any key

WeissP, 2019-09-28:
I want Joycon be an additional Input device in Linux, but not for game. So I don't really know if the drive is designed for this.

In the shell seems good:
[image]

[image]

but with command `xev` I can't see any key in Joycon. And the package "pynput" , which can monitor the keyboard in Python, does also not work.

EmeraldSnorlax, 2020-02-28:
Same issue.
No debug output beyond this is generated (as shown here https://www.youtube.com/watch?v=lw6h8uZDAV8)

According to dolphin, no button input, let alone joystick registers.
Using Linux56, Manjaro

parkerlreed, 2020-02-28:
FWIW `xev` is for keyboard/mouse input. Not for joysticks.

EmeraldSnorlax, 2020-02-28:
This seems to be getting more wierd.
Steam controller settings works perfectly (joycons paired as a single) with the left joycon, but the right only registers presses and movements on the joystick about 10% of the time.