riking/joycon #65: jcdriver occupies two full cpu threads

hicsuntdragons, 2020-03-06:
Hello,
I am currently using the `joycon-git` package from AUR, on Arch Linux 5.5.1
When I pair up the joycons using the root command, my CPU usage increases drastically.
Before pairing:
[image]
After pairing:
[image]
In fact, the cpu usage increases as soon as i run the program. Is this expected? Because it looks excessive.
My CPU is a Intel i5-7200U @ 3.100GHz, nothing impressive, but I assume it should handle this application.
Any ideas?

riking, 2020-03-09:
Uh oh. I don't have time to work on this program right now unfortunately, and also I have a steam controller now making this project less of a necessity.

Tetsujinfr, 2020-06-21:
do not have this problem on jetson xavier NX, CPU usage keeps being low before/after library is running. fyi