riking/joycon #66: Joycon timeout

OoLunar, 2020-05-07:
Hello!
I'm currently trying to connect my joycons to my Ubuntu 20.04 dist, however they keep timing out and disconnecting. Here's my exact steps:
```
- Go onto blueman-manager
- Hold sync button for 3 seconds on left joycon
- Connect with Joy-Con L
- Hold sync button for 3 seconds on right joycon
- Connect with Joy-Con R
- Go onto gnome-terminal
- `cd ~/go/bin`
- `sudo ./jcdriver`
- `sync`
- Press R and L (Not ZR or ZL)
- `syncoff`
- `ls`
- `setHomeLights cl1 1`
- `setHomeLights clr 1`
Timed out.
```
Please note that the `setHomeLights` or `setPlayerLights` command doesn't seem to have an affect on the controllers. Also, when syncing up with the controllers, I usually get the error
```
Error: context deadline exceeded
```
Twice per controller. I'm not sure what else could be going wrong. I'm using the latest git commit to compile the binary. Any clues?

UsernameNotAvalible, 2020-11-03:
The sethomelights dont do anything for me either. You did have a spelling mistake in your commands. You typed
```
- `setHomeLights cl1 1`
- `setHomeLights clr 1`
```
When you need to type
```
- `setHomeLights c1l 1`
- `setHomeLights c1r 1`
```
You had put "CL1". The correct way is "C1L". You had then put "CLR" which needs to be "C1R". that might be what is causing your problem. Also, rather than doing "cd ~/go/bin" then "./jcdriver", you can just do "cd ~/DirWhereYouDidTheInstall" then "./jcdriver" if you followed the normal instructions. Hope this helps!

OoLunar, 2020-11-03:
Whew boy this was awhile ago. Chances are the issue has been fixed, but I'll need to give it another run to see if everything is working now. Thanks for pointing out my issue @UsernameNotAvalible! It's very likely that was the problem.