riking/joycon #67: How to run jcdriver in the background at boot?

tidanneels, 2020-06-16:
I am trying to start jcdriver in background during the boot of my raspberry pi 3 on raspbian/retropie.
I tried /etc/init.d but the process stops automatically, same problem when I create a service with systemctl.
I am a beginner in Linux command line, can someone help me ?

maxerbox, 2020-06-18:
dup https://github.com/riking/joycon/issues/60

tidanneels, 2020-06-18:
This issue is closed thanks to @maxerbox . Here is my specific jcdriver.service file: 

`[Unit]
Description=Joycon driver
After=network.target

[Service]
ExecStart=sudo /home/pi/go/bin/jcdriver
WorkingDirectory=/home/pi
StandardInput=tty-force
TTYVHangup=yes
TTYPath=/dev/tty20
TTYReset=yes
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target
`
These 4 lines solved the problem: 
 `StandardInput=tty-force
TTYVHangup=yes
TTYPath=/dev/tty20
TTYReset=yes`

closed 2020-06-18 by tidanneels