riking/joycon #77: Problems building and installing with GO

Steven11q, 2022-08-01:
There already exist a couple of issues already detailing how the commands used for compiling are now depreciated. 
Is there any way to compile this as of right now?

nickprice101, 2023-01-07:
For anyone interested I got it compiled on my Pi4 running Bullseye by manually installing Go v1.19 (google go_installer.sh). Then I had too change the CC env to use the gcc armhf compiler (otherwise it uses -marm). I set this using "go env -w CC=arm-linux-gnueabihf-gcc". Obviously make sure it is installed. I also had to install libudev-dev:armhf.

ZucchelliDaniele, 2023-12-06:
Hi i Tried to compile it also and during the compile by using the command: 
* go install github.com/riking/joycon/prog4/jcdriver

# I got en error: 
 runtime/cgo
In file included from _cgo_export.c:4:
cgo-gcc-export-header-prolog:31:14: error: size of array ‘_check_for_64_bit_pointer_matching_GoInt’ is negative
# What i done/have
* Rpi4 running Bullseye
* Go v1.19 using go_installer.sh
* changed the compiler by the command go env -w CC=arm-linux-gnueabihf-gcc
* Installed the compiler arm-linux-gnueabihf-gcc by the command: sudo apt-get install gcc-arm-linux-gnueabihf
* Installed libudev-dev:armhf